tenbaht/sduino #99: Problem working with SPL

CircuitDigest, 2020-06-27:
Hey I find sduino as a great alternative for STVD and cosmic C compiler. Mainly because of the familiar Arduino IDE and easy code upload and serial monitor debugging option. So my plan is to use the IDE and compile SPL directly. I do not intend to use the existing Arduino functions or libraries I am only interested in the IDE and SPL. 

I wrote a simple code to control GPIO pins, the code compiled and uploaded successfully but I was not able to get it working on the hardware. The same code (equivalent), when compiled on cosmic C, works fine. The code I used is as follows 
`#define Green_LED GPIOA, GPIO_PIN_3
void setup() {
GPIO_DeInit(GPIOA); //prepare Port A for working 
GPIO_DeInit(GPIOB); // prepare Port B for working 
 
//Declare PA2  as input pull up pin
GPIO_Init (GPIOA, GPIO_PIN_2, GPIO_MODE_IN_PU_IT); 
 
//Declare PA3 as Push Pull Output pin
GPIO_Init (Green_LED, GPIO_MODE_OUT_PP_LOW_SLOW);
 
//Declare PB5 as push pull Output pin
GPIO_Init (GPIOB, GPIO_PIN_5, GPIO_MODE_OUT_PP_LOW_SLOW);
}

void loop() {
if (GPIO_ReadInputPin(GPIOA, GPIO_PIN_2)) //if button pressed
GPIO_WriteLow(Green_LED); //LED ON
else
GPIO_WriteHigh(Green_LED); //LED OFF
GPIO_WriteReverse(GPIOB,GPIO_PIN_5);
delay (100);
}
`

I used the same code for Cosmic C and STVD, more info on that [https://circuitdigest.com/microcontroller-projects/gpio-functions-on-stm8s-using-cosmic-c-and-spl-blinking-led-with-push-button](url)

So what am I missing here? My best guess is something has messed up the clock settings?

stefaandesmet2003, 2021-01-30:
you activate external interrupts on PA2, and use the default setting in the EXTI_CR1 register which is 00: falling edge and low level.
So as soon as you pull PA2 low, the ISR for external ints gets continuously executed until you release the button

shrikant-lahase, 2021-04-09:
> you activate external interrupts on PA2, and use the default setting in the EXTI_CR1 register which is 00: falling edge and low level.
> So as soon as you pull PA2 low, the ISR for external ints gets continuously executed until you release the button

How to activate external interrupts on PA2 on stm8s103f3 ? Please help me.