tenbaht/sduino #101: STM8S001 is locked and cannot be progammed anymore

YoYo1846, 2020-07-11:
Hello,

I'm prototyping around the low cost 8 pins stm8s001 / stm8l001
I'm using a stm8s103 board definition in arduino, since it's the same family.

This chip is different than the other ones in the familly since it doesn't have any RST pin to start the programming thru the SWIM pin.

Unfortunately, in your code (wiring-init.c), the first thing done is disabling this pin, which lock forever the chip from beeing programmed :

-----
void init()
{
#ifndef ENABLE_SWIM
	// free the SWIM pin to be used as a general I/O-Pin
	CFG->GCR = CFG_GCR_SWD;
#endif
[...]
-----

Of course, it could be possible to disable this code using the CFLAG ENABLE_SWIM but I'm not sure it is possible in arduino

Are you aware of this limitation for the stm8s001 / stm8l001 and therefore, is there a plan to change the default configuration in wiring-init.c ?

Thanks,
YoYo

YoYo1846, 2020-07-11:
Recommandations : 

Recommendation for SWIM pin (pin #8) sharing
As the NRST pin is not available on this device, if the SWIM pin should be used with the I/O pin functionality, it is recommended to add a ~5 seconds delay in the firmware before changing the functionality on the pin with SWIM functions. This action allows the user to set the device into SWIM mode after the device power on and to be able to reprogram the device. If the pin with SWIM functionality is set to I/O mode immediately after the device reset, the device is unable to connect through the SWIM interface

smartswitchsio, 2020-10-29:
Kindly add stm8s001 in board section

thinhx2, 2020-11-19:
[image]
Try this one

rtek1000, 2023-03-26:
Hello,

I already had similar problem with PIC12F675.

I left the WDT with a very short time, and also disabled the RESET.

But with a lot of effort I managed to reprogram this PIC, making pulses on the power line, to force the RESET synchronized with the programmer.

I don't know if this can solve the STM8S001 problem, but I leave here the report for those who use PIC at least.

I am trying to disable SWIM as it is the only free pin on the W1209 thermostat board (STM8S003), and it would be nice to have some way to export the data.

rtek1000, 2023-03-26:
I found this:

Pg 58:
> HVP
> If this option is checked, the programmer uses the “high”
> voltage at pin P for communication with the device.
> This allows you to program a device with the external
> RESET signal switched off

Pg 57:
> Programming method
> • HVP Traditional programming will be applied using a voltage of 8 V to 13 V at pin P.

Pg 9:
> HVP (High Voltage Programming) is a
> programming mode in which a higher
> voltage than the power-supply voltage is
> applied to pin P in the initial phase.

[Settings Associated with AVR and 8051 Microcontrollers](https://www.kanda.com/datasheet/14-09-2015-1226forte_en.pdf)

Did ST put this option in STM8S001?