tenbaht/sduino #103: stm8flash: The following device has no driver … Could not open USB device

spth, 2020-08-05:
I've tried to use the stm8flash.exe from the 0.5 release as standalone program from the command line on Windows 10 with an STM8AF board (which has an integrated stlinkv2).

But I get an error (doesn't matter if I'm admin):

~~~~
libusb: info [windows_get_device_list]: The following device has no driver […]
[…]
Could not open USB device.
~~~~

I've read something about a WinUSB driver that libusb might need, but don't know hot to install it for the stlinkv2.

tenbaht, 2020-10-15:
Hmm... I don't have a Windows 10 system around for testing. But the repro for libusb mentions a driver issue for USB 3 ports with a NEC chipset: https://github.com/libusb/libusb/wiki/Windows
It would be great to hear if you got the problem solved.!

spth, 2020-10-15:
I don't really use Windows 10 either, just did a quick test back then.