tenbaht/sduino #107: [Linux] [STM8S103F3 Breakout Board] Could not open USB device

wingedrhino, 2020-08-19:
Hi,

I get the following error when I try to upload a sketch on my STM8:

```
/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/wingedrhino/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/wingedrhino/.arduino15/packages -libraries /home/wingedrhino/Arduino/libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10813 -build-path /tmp/arduino_build_6552 -warnings=all -build-cache /tmp/arduino_cache_516551 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM8Tools.path=/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05 -prefs=runtime.tools.STM8Tools-2019.02.05.path=/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05 -prefs=runtime.tools.sdcc.path=/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242 -prefs=runtime.tools.sdcc-build.11242.path=/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242 -verbose /usr/share/arduino/examples/01.Basics/Blink/Blink.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/wingedrhino/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/wingedrhino/.arduino15/packages -libraries /home/wingedrhino/Arduino/libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10813 -build-path /tmp/arduino_build_6552 -warnings=all -build-cache /tmp/arduino_cache_516551 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM8Tools.path=/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05 -prefs=runtime.tools.STM8Tools-2019.02.05.path=/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05 -prefs=runtime.tools.sdcc.path=/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242 -prefs=runtime.tools.sdcc-build.11242.path=/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242 -verbose /usr/share/arduino/examples/01.Basics/Blink/Blink.ino
Using board 'stm8sblue' from platform in folder: /home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0
Using core 'sduino' from platform in folder: /home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0
Detecting libraries used...
/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdcc.sh /home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc /tmp/arduino_build_6552/sketch/Blink.ino.cpp /dev/null re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdcc.sh /home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc /tmp/arduino_build_6552/sketch/Blink.ino.cpp /tmp/arduino_build_6552/preproc/ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include -DARDUINO_LIB_DISCOVERY_PHASE
/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc /tmp/arduino_build_6552/sketch/Blink.ino.cpp /tmp/arduino_build_6552/preproc/ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include -DARDUINO_LIB_DISCOVERY_PHASE
Mark re12:/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include -DARDUINO_LIB_DISCOVERY_PHASE /tmp/arduino_build_6552/sketch/Blink.ino.cpp -o /tmp/arduino_build_6552/preproc/ctags_target_for_gcc_minus_e.cpp
cpp gefunden
/usr/bin/arduino-ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_6552/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/wingedrhino/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdcc.sh /home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc /tmp/arduino_build_6552/sketch/Blink.ino.cpp /tmp/arduino_build_6552/sketch/Blink.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -Wp-Wall -V -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include
/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc /tmp/arduino_build_6552/sketch/Blink.ino.cpp /tmp/arduino_build_6552/sketch/Blink.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -Wp-Wall -V -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include
Mark re2:/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -Wp-Wall -V -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include /tmp/arduino_build_6552/sketch/Blink.ino.cpp -o /tmp/arduino_build_6552/sketch/Blink.ino.cpp.o
cpp gefunden
+ /home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcpp -nostdinc -Wall -std=c11 -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -Wall -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/wingedrhino/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/wingedrhino/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include -obj-ext=.rel -MMD /tmp/arduino_build_6552/sketch/Blink.ino.cpp.d -MT /tmp/arduino_build_6552/sketch/Blink.ino.cpp.o -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_MODEL_MEDIUM -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCC= [... cut ...]
At first I thought it's my crappy ST-Link V2 (fake) dongle - the default blink program wasn't always running when I connecrted 3.3V, SWM, GND, and NRST pins respectively to 3.3V, SWIM, GND, and RST. But I managed to get it working when I removed the 3.3V connection, and powered the board through an external USB cable (I see a blinking red LED and a steady blue one).

Just to be safe, I also added udev rules for stlink, following the instructions here: https://www.codementor.io/@hbendali/getting-started-with-stm8-development-tools-on-gnu-linux-zu59yo35x#usb-troubleshooting

But I am still unable to upload anything to the microcontroller.

Any other tips on debugging this issue?

DangerPie, 2020-09-01:
There is boards with disconnected GND - did you checked that?

wingedrhino, 2020-09-01:
> There is boards with disconnected GND - did you checked that?

Wow disconnected GND? Where? You mean between the ground on the parallel legs and the ground on the programming pins tail?

DangerPie, 2020-09-01:
Yes: https://youtu.be/5Rbz5ZbxxB4?t=155

tenbaht, 2020-10-15:
Are you still having problems with the flash tool? The description definitely sounds like a permission issue. Are you member of the plugdev group? (check with `id`) And did you unplug/replug the stlink after changing and reloading the udev rules?

wingedrhino, 2020-10-16:
Ahoy!

I stopped attempting to use that breakout board. I shall try again this weekend, with the grounds connected manually!

tenbaht, 2020-10-29:
I added an FAQ page and mentioned the faulty GND connection for stm8blue boards.
https://tenbaht.github.io/sduino/usage/faq/
If you still have problems with your board, please re-open this issue.

closed 2020-10-29 by tenbaht