tenbaht/sduino #109: STM8S

laxman432, 2020-09-21:
HI WE LIKE TO USE STM8SF3P6 OUR APPLICATION WITH GSM COMMUNICATION SO WE LIKE TO USE SOFTWARE SERIAL FOR THAT AND FOR SERIAL CHECKING WE LIKE USE SERIAL PORT  IS MY CONCEPT WORKOUT ? IS IT SUPPORT SOFTSERIAL ?

mcer12, 2020-10-06:
@laxman432 You might want to check out STM32F030 instead. Slightly more expensive but fully compatible with arduino ide.

laxman432, 2020-10-06:
ALREADY WE DONE WITH STM32 BLUE PILL BUT WE ARE LOOKING FOR LOW COST CONTROLLER IF ANY SUGGEST ME IT SHOULD SUPPORT ARDUINO PROGRAMING

mcer12, 2020-10-06:
I just suggested you one...

tenbaht, 2020-10-13:
I am not sure which MCU you mean by STM8SF3P6. But sduino does not support SoftwareSerial (yet). There is HardwareSerial which you could use.
An mcer12 is right: If comprehensive library support is important for your project the STM8 might not be the ideal choice. Have you considered an ATtiny? Full Arduino support, but cheaper than a full ATmega.

closed 2020-10-13 by tenbaht

laxman432, 2020-10-14:
Suggest me best in attiny controller like stm8sf3p6  it should support
serial communication only and it should support Arduino ide