tenbaht/sduino #110: SPI library

kriswiner, 2020-10-01:
On the README frontpage is mentioned hardware SPI but I don;t see an SPI.h/SPI.cc library in the latest release. Is such a library necessary (I would think so)? And do you have an example of using hardware SPI with the S103F? Thanks...

tenbaht, 2020-10-13:
The SPI library is part of the core files in sduino/stm8/cores/sduino
A simple `#include "SPI.h"` is enough to use it.
And you are right, there is no proper SPI example except for the one you already found. I am open for suggestions. Is there any common SPI device that could be used for a demonstration?

kriswiner, 2020-10-13:
PMW3901
<https://www.tindie.com/products/onehorse/pmw3901-optical-flow-sensor/> is
a good SPI sensor for an example. Once I get my boards from China I will
give it a try.

closed 2020-10-13 by tenbaht

tenbaht, 2020-10-13:
Oh, nice sensor. But not very common, I am afraid. The most common thing I can think of are displays like the classic Nokia 5110. But they require quite complex communication before they can do anything useful - not so good for a compact example.
The PCD8544 library includes some examples, but it uses bitbanged SPI instead of the real thing.

kriswiner, 2020-10-13:
The PMW3901 is used by bitcraze in their CrazyFlie offering, and both
Sparkfun and Adafruit have breakout I believe. So not too uncommon...