tenbaht/sduino #118: Fails to compile when using Ucglib (ST7735 & other OLEDs)

IceC00l, 2021-02-05:
Trying to compile any Example code (Hello World) when using Ucglib yields a fatal error: invalid argument

Documents/Arduino/libraries/Ucglib/src/Ucglib.h:65: syntax error: token -> 'Ucglib' ; column 12

This is where "class Ucglib : public Print" is in the header file. 

Is there a chance this ST7735 compatible library can be supported/fixed for this MCU family?

prosper00, 2021-02-05:
that's a c++ lib you're trying to include. sduino is plain C only.

However, it looks like there's a plain C version of that lib on github. I'd start by trying that one: https://github.com/olikraus/ucglib/tree/master/csrc

IceC00l, 2021-02-06:
Right - thanks for the heads up: currently I have the ucglib installed already and I see no csrc directory under the library (which is normal I guess - I am using it for other ATmel stuff), if I manually include that directory I suppose I need to modify some environment vars to use that one instead of the C++ src dir to try to compile with your toolchain or do I have to do some porting as explained in https://tenbaht.github.io/sduino/api/migration/? Thanks again!