tenbaht/sduino #122: add support for Nucleo Boards

gicking, 2021-03-07:
please consider adding support for the Nucleo-8S208RB & STM8S207S8 development board by STM. Thanks a lot in advance!

https://www.st.com/en/evaluation-tools/nucleo-8s208rb.html

https://www.st.com/en/evaluation-tools/nucleo-8s207k8.html

mcer12, 2021-03-30:
@gicking check this out. At least one of the boards seems to be compatible

mcer12, 2021-03-30:
https://github.com/stm32duino/Arduino_Core_STM8

gicking, 2021-03-31:
@mcer12 thanks for pointing this out! However:
- that's a different project (mainly) aiming at STM32
- uses Cosmic compiler and is Windows only

Therefore I would like to keep this issue open

mcer12, 2021-03-31:
Yes it is, but it’s an option. This repo hasn’t been updated since last year.

closed 2021-04-23 by gicking