tenbaht/sduino #124: tone() support or workaround?

HACKER-3000, 2021-03-13:
I would really like to have the tone() function or some way to replicate it's functionality, if it's not too much of a hastle to implement.

maxgerhardt, 2021-03-22:
It has a tone generator, the "BEEPER" periperheral. That thing however only has 3 possible output frequencies: 1, 2 or 4 kHz (refer datasheet / user manual and e.g. [here](http://embedded-lab.com/blog/starting-stm8-microcontrollers/11/)). So it can't support arbitrary audible frequencies from like 0-44kHz with that.

But ofc it should be possible with a timer / PWM output. I've outlined some initial ideas some time ago [there](https://community.platformio.org/t/stm8s-compiling-or-even-just-setting-up-to-work-vscode-help/18663/9?u=maxgerhardt).

However a simple high-low write and a microsecond delay works on my STM8S103F3 (and pretty much every other). I've slightly adapted a sketch for [a tetris theme](https://github.com/maxgerhardt/TetrisThemeArduino/blob/master/src/TetrisThemeArduino.c) and it plays nicely on a buzzer, with simulated 2-tone polyphony.

stefaandesmet2003, 2021-04-12:
for what it's worth, I added an implementation here : https://github.com/stefaandesmet2003/sduino
It's using TIM2 (similar to AVR), but STM8 TIM2 is 16-bit.
STM8S003 has TIM2 peripheral, so it should work too.

HACKER-3000, 2021-04-14:
very cool! you can open a pull request

stefaandesmet2003, 2021-04-14:
agree, but it looks like this repo is a bit stalled for the moment