tenbaht/sduino #125: TM1637 support

sh4had, 2021-04-03:
I would like to use a 7 segment display on tm1637 in my sketch, there is already a library for stm8s and SDCC  [https://github.com/indigo6alpha/stm8s-tm1637](url) ,but I can't figure out how to use it in the arduino ide (sduino). Or how to safely connect such a display without libraries.

prosper00, 2021-04-03:
In general, put the source files into your project directory, and put #include "whatever.h" in your sketch

sh4had, 2021-04-04:
Oh no, I think it just won't work, it's not under arduino. I tried of course, added #include <main.c> and got errors:

```
 Arduino errors: 1.8.13 (Windows 10), Board: "STM8S103F3 Breakout Board"
C:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\tm1637stm8s.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include -DARDUINO_LIB_DISCOVERY_PHASE
Mark re12:C:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include -DARDUINO_LIB_DISCOVERY_PHASE sketch\tm1637stm8s.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\tm1637stm8s.ino.cpp sketch\tm1637stm8s.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\admin\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\tm1637stm8s.ino.cpp -o sketch\tm1637stm8s.ino.cpp.o
cpp gefunden
C:/Users/admin/AppData/Local/Arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino/main.c:27: error 98: conflict with previous declaration of 'main' for attribute 'type' at sketch\tm1637stm8s.ino.c:1
from type 'void function ( ) fixed'
to type 'int function ( ) fixed'
exit status 1
Compilation error for the STM8S103F3 Breakout Board.
```

I actually do not insist on using this library, you need any way to achieve the connection of this display.

sh4had, 2021-04-08:
https://github.com/bitbank2/tm1637/tree/master/Arduino  - works great!
[image]

closed 2021-04-08 by bamboo-master

Sidbadode, 2022-08-15:
It Shows error

Sidbadode, 2022-08-15:
It Shows error
[image]

jay-ramani, 2022-08-15:
> It Shows error [image]

You have an error in code - multiple definitions of _dummy_variable

[image]

Sidbadode, 2022-08-16:
I used the same given code 

#include "tm1637.h"

//
// Larry's TM1637 Library Demo
// Displays counter from 0000 to 9999 with flashing colon
//
void setup() {
  // put your setup code here, to run once:
  tm1637Init(8, 9); // clock, data pins
  tm1637SetBrightness(4);
}

void loop() {
int i;
char szTemp[8];

  // put your main code here, to run repeatedly:
  for (i=0; i<9999; i++)
  {
      if (i & 1)
         sprintf(szTemp, "%02d:%02d", i/100, i% 100);
      else
         sprintf(szTemp, "%02d %02d", i/100, i % 100);
    tm1637ShowDigits(szTemp);
    delay(500);
  }
}

Sidbadode, 2022-08-16:
if i only use the library then also shows the same error
[image]

bamboo-master, 2022-08-16:
I remember that it was necessary to rename some file so that all this would work in the arduino ide. Keep the working version, you need to unpack it to the folder `%SYSTEMDRIVE%\Users\%username%\Documents\Arduino\libraries` 
[tm1637.zip](https://github.com/tenbaht/sduino/files/9351653/tm1637.zip)

Sidbadode, 2022-08-17:
Thank you so much yes it works .
[image]