tenbaht/sduino #127: Using Timer and interrupt for decoding IR signal timings on STM8 using sduino

shrikant-lahase, 2021-04-09:
Hi Tenbaht,

I am new to this environment. I am trying to do get IR signal and measure timings of IR signal. For that i want to use timer and external interrupt.  Please give any example code or starting point so that I can code for this on arduino IDE. 

Thanks

stefaandesmet2003, 2021-04-11:
the compiler tells you the EXTI_PORTA_IRQHandler is already defined by the sduino library. Have a look at WInterrupts.c and you will understand how to integrate your ISR. 
The EXTI register can only be written with global interrupts disabled -> check your setup()
You also need to enable the external interrupt at the port (GPIOx->CR2 register)
Working example below.
Hope this helps.

```
// test pin change interrupt on pin PC3
// the pin level on PC3 is copied onto the built-in led 

static void portc_irq (void) {
  digitalWrite(LED_BUILTIN,digitalRead(PC3));
}

void setup() {
  pinMode(LED_BUILTIN,OUTPUT);
  digitalWrite(LED_BUILTIN,LOW); //led on

  pinMode (PC3, INPUT_PULLUP);
  attachInterrupt(2,portc_irq, 0); // mode is not implemented in sduino, so we do the config manually after

  disableInterrupts(); // EXTI->CR1 only writable under disabled interrupts (CCR=3)
  EXTI->CR1 = 0x30; // set falling+rising interrupt for all pins on port C
  GPIOC->CR2 = 0x08; // enable ext interrupt on pin PC3
  enableInterrupts();
}

void loop() {
}

```

shrikant-lahase, 2021-04-12:
> the compiler tells you the EXTI_PORTA_IRQHandler is already defined by the sduino library. Have a look at WInterrupts.c and you will understand how to integrate your ISR.
> The EXTI register can only [...]

Thank you very much. It worked.

shrikant-lahase, 2021-04-12:
Can you please similar example for  timer4 in counter mode ?

shrikant-lahase, 2021-04-12:
Thank you very much for a nice explanation. It worked.

Please give any similar example for a timer counter of stm8s103f3p

stefaandesmet2003, 2021-04-12:
TIM4 is in use for millis() / micros()
Have a look at wiring-millis.c

shrikant-lahase, 2021-04-12:
Where is this file  `wiring-millis.c` available ?

I added code as follows:
```
  /* TIM4 configuration:
   - TIM4CLK is set to 16 MHz, the TIM4 Prescaler is equal to 128 so the TIM1 counter
    clock used is 16 MHz / 128 = 125 000 Hz
   - With 125 000 Hz we can generate time base:
       max time base is 2.048 ms if TIM4_PERIOD = 255 --> (255 + 1) / 125000 = 2.048 ms
       min time base is 0.016 ms if TIM4_PERIOD = 1   --> (  1 + 1) / 125000 = 0.016 ms
   - In this example we need to generate a time base equal to 1 ms
    so TIM4_PERIOD = (0.001 * 125000 - 1) = 124 
    */

uint32_t timercount=0;
uint32_t timer_1secCnt = 0;
bool measuring = false;
uint8_t capture[20] = {0};
unsigned int i=0;
unsigned int j=0;
bool flag = false;

INTERRUPT_HANDLER(TIM4_UPD_OVF_IRQHandler,23)
{
  timercount++;  
  TIM4_ClearFlag(TIM4_FLAG_UPDATE); //very important, if not cleared, the interrupt function is called over and over stalling the uP
  //Serial_println_i(timercount);
  if(timercount == 500){
     timer_1secCnt++;
     Serial_println_i(timer_1secCnt);
     timercount = 0;
  }
  flag = true;
}

// Handle interrupt
static void portc_irq (void) {
  int buttonState = digitalRead(PC3);
  digitalWrite(LED_BUILTIN,buttonState);  
  Serial_print_s(">>"); Serial_println_i(timer_1secCnt);
}

void setup() {
  pinMode(LED_BUILTIN,OUTPUT);
  digitalWrite(LED_BUILTIN,LOW); //led on
  pinMode (PC3, INPUT_PULLUP);
  attachInterrupt(2,portc_irq, 0); // mode is not implemented in sduino, so we do the config manually after

  disableInterrupts(); // EXTI->CR1 only writable under disabled interrupts (CCR=3)
  EXTI->CR1 = 0x30; // set falling+rising interrupt for all pins on port C
  GPIOC->CR2 = 0x08; // enable ext interrupt on pin PC3
  
    CLK_HSIPrescalerConfig(CLK_PRESCALER_HSIDIV1);
    TIM4_DeInit();
    //128 frequency division x=16M/128 interrupt overflow=x/0xfa =250 enter the interrupt once for 2ms.  For 1ms TIM4ARR=124.
    TIM4_TimeBaseInit(TIM4_PRESCALER_128, 250); 
    TIM4_ClearFlag(TIM4_FLAG_UPDATE);
    TIM4_ITConfig(TIM4_IT_UPDATE, ENABLE);
    TIM4_Cmd(ENABLE);
  
  enableInterrupts();
  Serial_begin(115200);  //init uart
}

void loop() {

 if(flag){
 // Serial_println_i(timercount);
  flag=false;
 }
}
```

shrikant-lahase, 2021-04-12:
Thanks, I'll check it out.  Where is this file wiring-millis.c available ?

stefaandesmet2003, 2021-04-12:
you must be mistaking this issue page for stackoverflow ;)

neoblack2, 2021-04-16:
> the compiler tells you the EXTI_PORTA_IRQHandler is already defined by the sduino library. Have a look at WInterrupts.c and you will understand how to integrate your ISR.
> The EXTI register can only [...]

Dont work on STM8S105K6.

neoblack2, 2021-04-16:
Is it so hard for the author to make working code with interrupts? Moreover, there is an example, but not working.

stefaandesmet2003, 2021-04-16:
for me the code works on STM8S103.
I don't have a STM8S105, but from the datasheet the register map looks identical for what concerns this example.
"dont work" isn't very precise, if you need useful help :)