tenbaht/sduino #130: unable to install in arduino 2 beta

dickshaydle, 2021-05-22:
I cannot Install sduino on arduino 2 beta. On v1.8.13 it works to install it.

Error Message:
```
Downloading packages
sduino:sdcc@build.11242
sduino:stm8@0.5
Installing sduino:STM8Tools@2019.02.05
sduino:STM8Tools@2019.02.05 installed
Installing sduino:sdcc@build.11242
sduino:sdcc@build.11242 installed
Installing sduino:stm8@0.5
Failed to install platform: sduino:stm8.
Error: 2 UNKNOWN: installing platform sduino:stm8@0.5: searching package root dir: no unique root dir in archive, found '/home/user/.arduino15/tmp/package-636989126/STM8S_StdPeriph_Driver' and '/home/user/.arduino15/tmp/package-636989126/cores'
```

Arduino Version:
```
Version: 2.0.0-beta.6
Date: 2021-05-12T10:22:15.408Z
CLI Version: 0.18.2 alpha [7b5a22a4]
```

jay-ramani, 2022-03-27:
+1. I've run into the same issue. Looked up the issue and found something similar in #https://github.com/arduino/arduino-cli/issues/325. From the discussion, looks like it needs a fix in the package.

Would be great to see a resolution to this. Thanks in advance for the attention.

P. S: Did anyone manage to figure a workaround, given that the bus was reported on 22nd May 2021?

Kturner05, 2024-04-22:
Still not working as of May 2024.

platima, 2025-05-16:
Use this, it seems: https://github.com/akashkapashia/stm8_arduino