tenbaht/sduino #132: Any chance for STM7?

SanZamoyski, 2021-09-20:
Hi!

I got into my hands over 10 boards with STM7 (72C334). I think it might be good for some of my projects. 
Generally, I'm using Arduino IDE ALOT, so it would be awesome to use it to program those boards. 
I'm familiar with ISP etc. 
As far as I know, STM8 is extended STM7. That's why I'm asking here.

Best regards!

SanZamoyski, 2021-11-29:
How about this one:
https://sourceforge.net/projects/sdccstm/ ?

SanZamoyski, 2021-11-29:
Yes, I've seen that. But let's be honest - it's an old mcu. Probably, there is very little chance for anything better.
The question is: is there a lot of work to extend this to STM7?

roybaer, 2021-11-29:
Probably a lot more than you would want to invest.

SanZamoyski, 2021-11-30:
Can You explain a little?

roybaer, 2021-11-30:
Well, for starters you would at least need the entire hardware abstraction code for the peripherals, i.e. you would actually have to rewrite all of the libraries or at least significant parts thereof.
Then you need board support. Since there are definitely no ST7 boards already in the code base, you would need board descriptions for your hardware.
You would also need a way to program it. Stm8flash and stm8gal are – big surprise – for STM8, only.

All of this assumes the best-case scenario, i.e. that the old SDCC fork for ST7 actually works, does so reliably, produces adequate code and supports all relevant features. I am not so sure about that.

Incidentally, you may have noticed that I am not the maintainer of this repository but only an occasional contributor.
This project is not particularly active, so if you want new functionality, you will basically have to add it yourself.