tenbaht/sduino #137: STM8S103 with arduino IDE has error

TanveerHussain123, 2022-01-28:
Did a board manager install of Sduino per the instructions (running Arduino 1.8.19):
Start the Arduino-IDE. In File->Preferences, Settings tab, enter

https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json

as an Additional Boards Manager URL.

Open Tools->Board:...->Boards Manager
Find Sduino by typing 'sd' into the search line
Click on the list entry
Click on Install.

blink LED program is working but some program gives error for example serial communication code and ultrasonic code
after that getting this error

"Error compiling for board STM8S103F3 Breakout Board".

complete error is below.
:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\HCSR04.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\tanve\Documents\Arduino\libraries\HCSR04_ultrasonic_sensor\src -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re12:C:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\tanve\Documents\Arduino\libraries\HCSR04_ultrasonic_sensor\src -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\HCSR04.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\HCSR04.ino.cpp sketch\HCSR04.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\tanve\Documents\Arduino\libraries\HCSR04_ultrasonic_sensor\src -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\tanve\Documents\Arduino\libraries\HCSR04_ultrasonic_sensor\src -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\tanve\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\HCSR04.ino.cpp -o sketch\HCSR04.ino.cpp.o
cpp gefunden
C:/Users/tanve/Documents/Arduino/libraries/HCSR04_ultrasonic_sensor/src/HCSR04.h:4: syntax error: token -> 'HCSR04' ; column 12
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

prosper00, 2022-01-28:
the HCSR04 library you're trying to use is a C++ library. sduino is standard-C only.

TanveerHussain123, 2022-01-29:
can i conver this library in to C, or any solution if i want to use this library, any recomendation. thanks in advance.

prosper00, 2022-01-29:
No. You'd have to rewrite it. Or use an arduino nano

semibank, 2024-04-01:
Is sample code for using the sr04 ultrasonic sensor supported?