tenbaht/sduino #147: Sleep and interrupt in Sduino

chirag-uAtech, 2023-02-17:
Hi community,

I am new to Sduino and Stm8s. Have some prior experience working with Arduino, AVRs & STM32. I chose STM8s due to its lower price for a project. After developing the project, I need to reduce the current usage as much as possible to make the system Standby Mode. I have tried and failed while using Interrupt with a button.

For the sake of clear understanding, I want to develop a project that can **control LED using a Push Button**. I want the system to be in **sleep mode for the rest of the time**.

The sduino repo contains an example code for interrupt, that is:

```
#define BUTTON	PA2

volatile uint8_t flag = 0;

void on_button_pressed(void)
{
	flag = 1;
}
void setup()
{
	pinMode(LED_BUILTIN, OUTPUT);
	digitalWrite(LED_BUILTIN, 1);	// turn off the LED

	pinMode(BUTTON, INPUT_PULLUP);

	attachInterrupt(digitalPinToInterrupt(BUTTON), on_button_pressed, FALLING);
}
void loop()
{
	if (flag) {
		digitalWrite(LED_BUILTIN, 0);
		delay(300);
		digitalWrite(LED_BUILTIN, 1);
		flag = 0;
	}
}

```
The code does not get compiled in Arduino. On searching the internet for a bit, I found changing _digitalPinToInterrupt(pin);_ to _digitalPinToPort(pin);_ may work. The code gets compiled now but the result remains unexpected.

After digging some more into the problem, I found one more possible solution:

  ```
GPIO_Init(GPIOA, GPIO_PIN_3, GPIO_MODE_IN_FL_IT);
  disableInterrupts();
  EXTI_SetExtIntSensitivity( EXTI_PORT_GPIOA, EXTI_SENSITIVITY_RISE_ONLY);  
  enableInterrupts();
  attachInterrupt(INT_PORTA & 0xFF,ISR,0);
```

Source: https://github.com/tenbaht/sduino/issues/92

I tried this as well in my code. But still unable to make it work.

My requirement is to attachInterrupt on pin PA2 & use it to wake from sleep. Similar issues have been raised multiple times but not resolved completely.

Kindly provide some input on this problem.

MR-IC, 2023-12-05:
You can implement it simply by writing a command "wfi" it halts the microcontroller and waits for the interrupt to occur.

Before entering this command make sure that you have proper interrupt attached.

chirag-uAtech, 2024-01-08:
@MR-IC How do I do that? I am new to development on STM8 and can't find any function in the whole sduino library relating to this. It would help greatly if you can attach a simple code to go to sleep and then wake from button using EXTI.

g1ra, 2024-12-19:
Hi . I found wfi at https://github.com/tenbaht/sduino/blob/0ce7e4c381028f6472eb29e8459210cee90b937d/sduino/stm8/STM8S_StdPeriph_Driver/inc/stm8s.h#L2747 
Please help How can I set pin for interrup wfi and continue running the program ?