tenbaht/sduino #150: Package install error: searching package root dir: no unique root dir in archive

EstevanTH, 2023-08-30:
Hey there,

I had an issue while installing the platform `sduino:stm8`, in any version (the current last version being `0.5`.

Characteristics:

- OS: Windows 11 x86-64
- Arduino IDE 2.1.1

Installation failed with:

```cmd
C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build>arduino-cli core install "sduino:stm8@0.5" --additional-urls "https://raw.githubusercontent.com/tenbaht/sduino/master/package_sduino_stm8_index.json"
Tool sduino:STM8Tools@2019.02.05 already installed
Tool sduino:sdcc@build.11242 already installed
Téléchargement des paquets...
sduino:stm8@0.5 sduino:stm8@0.5 déjà téléchargé
Installing platform sduino:stm8@0.5...
Error during install: Cannot install platform: installing platform sduino:stm8@0.5: searching package root dir: no unique root dir in archive, found 'C:\Users\xxx\AppData\Local\Arduino15\tmp\package-4125649767\STM8S_StdPeriph_Driver' and 'C:\Users\xxx\AppData\Local\Arduino15\tmp\package-4125649767\cores'
```

Opening the archive with 7-zip, I saw inside that the root directory inside of the `tar` archive is named `.`. This misled the installation procedure.

[image]

So I rebuilt the archive, putting all the content inside of a single root directory `sduino-core-0.5`. I modified my local `package_sduino_stm8_index.json` to make the size and the SHA-256 correspond with the new values. Hurray, it works!

[image]

```cmd
C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build>arduino-cli core install "sduino:stm8@0.5" --additional-urls "package_sduino_stm8_index.json"
Tool sduino:STM8Tools@2019.02.05 already installed
Tool sduino:sdcc@build.11242 already installed
Téléchargement des paquets...
sduino:stm8@0.5 sduino:stm8@0.5 déjà téléchargé
Installing platform sduino:stm8@0.5...
Configuration de la plateforme....
Platform sduino:stm8@0.5 installed
```

Thank you!

Cheers ❤️

cintix, 2023-09-13:
I have the same issue on both my OSX and Windows 11

mf0o, 2023-11-29:
Thanks Estevan.

Facing the same issue on macOS and Arduino IDE 2.2.1.
Thanks to your precise description I could replicate the fix fast and installed the package successfully.

Thank you a lot!

paturiaadarsh, 2024-07-19:
> Hey there,
> 
> I had an issue while installing the platform `sduino:stm8`, in any version (the current last version being `0.5`).
> 
> Characteristics:
> 
> * OS: Windows 11 x86-64
> * Arduino IDE 2.1.1
> 
> Installation failed with:
> 
> ```batchfile
> C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build>arduino-cli core install "sduino:stm8@0.5" --additional-urls "https://raw.githubusercontent.com/tenbaht/sduino/master/package_sduino_stm8_index.json"
> Tool sduino:STM8Tools@2019.02.05 already installed
> Tool sduino:sdcc@build.11242 already installed
> Téléchargement des paquets...
> sduino:stm8@0.5 sduino:stm8@0.5 déjà téléchargé
> Installing platform sduino:stm8@0.5...
> Error during install: Cannot install platform: installing platform sduino:stm8@0.5: searching package root dir: no unique root dir in archive, found 'C:\Users\xxx\AppData\Local\Arduino15\tmp\package-4125649767\STM8S_StdPeriph_Driver' and 'C:\Users\xxx\AppData\Local\Arduino15\tmp\package-4125649767\cores'
> ```
> 
> Opening the archive with 7-zip, I saw inside that the root directory inside of the `tar` archive is named `.`. This misled the installation procedure.
> 
> ![image](https://private-user-images.githubusercontent.com/5159778/264330258-6008095a-373a-44b1-bb3f-13089c5a276b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzNzAxNjIsIm5iZiI6MTcyMTM2OTg2MiwicGF0aCI6Ii81MTU5Nzc4LzI2NDMzMDI1OC02MDA4MDk1YS0zNzNhLTQ0YjEtYmIzZi0xMzA4OWM1YTI3NmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTlUMDYxNzQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDQ3YWNmNDJhYWVlYTM4YTM0YWRlN2FmOWRjZWRlNDU3NmQzODBiYWQxMGQ3ZGI3OWVkODhkNzRkZmQ3ZGE5YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.40-QW-6gFPF2OvNiz4UYmXLBi3VRVyS5NbdSauGzRos)
> 
> So I rebuilt the archive, putting all the content inside of a single root directory `sduino-core-0.5`. I modified my local `package_sduino_stm8_index.json` to make the size and the SHA-256 correspond with the new values. Hurray, it works!
> 
> ![image](https://private-user-images.githubusercontent.com/5159778/264330547-2409fb27-e4fc-4b6f-9afa-52aec20727af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzNzAxNjIsIm5iZiI6MTcyMTM2OTg2MiwicGF0aCI6Ii81MTU5Nzc4LzI2NDMzMDU0Ny0yNDA5ZmIyNy1lNGZjLTRiNmYtOWFmYS01MmFlYzIwNzI3YWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTlUMDYxNzQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDdiYmEyMGZjZmJhZDhkN2ZmNzlkZTQwZjk5YzIyNzg3MzI2NmU2ODI5ZmJmNmZiNTg1YWM1NTI3ZWU2MWM1ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.BKKsRH_E9VAuUsmvUe7IVvajEUEDxdanljOI0PJ1f7E)
> 
> ```batchfile
> C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build>arduino-cli core install "sduino:stm8@0.5" --additional-urls "package_sduino_stm8_index.json"
> Tool sduino:STM8Tools@2019.02.05 already installed
> Tool sduino:sdcc@build.11242 already installed
> Téléchargement des paquets...
> sduino:stm8@0.5 sduino:stm8@0.5 déjà téléchargé
> Installing platform sduino:stm8@0.5...
> Configuration de la plateforme....
> Platform sduino:stm8@0.5 installed
> ```
> 
> Thank you!
> 
> Cheers ❤️

hey there, if u dont mind can u explain this thing breifly, i am having the same issue

Shoieb-Morad, 2024-10-05:
> hey there, if u dont mind can u explain this thing breifly, i am having the same issue

did you solve it?

Sod-Almighty, 2025-07-03:
Are you ever going to fix this bug?