tenbaht/sduino #155: Installing the Board in the Arduino is failing

akashkapashia, 2023-12-16:
Error: 13 INTERNAL: Cannot install platform: installing platform sduino:stm8@0.5-pre2: searching package root dir: no unique root dir in archive, found 'C:\Users\Akash\AppData\Local\Arduino15\tmp\package-2304909887\STM8S_StdPeriph_Driver' and 'C:\Users\Akash\AppData\Local\Arduino15\tmp\package-2304909887\cores'

This can be fixed by putting the release download in the release version.
means add a additional folder with version name "v5.0/downloaded"
[Uploading v0.5.zip…]()
 files

akashkapashia, 2023-12-17:
https://github.com/akashkapashia/stm8_arduino/blob/master/README.md

use this meanwhile

bullygram, 2024-06-11:
@akashkapashia still no luck, unable to install sduino!

currently using Arduino version 2.1.1. Any help will be greatly appreciated!

bullygram, 2024-06-11:
Got it fixed, I now have **two portable versions (downloaded as ZIP)** of the Arduino IDE, one is a legacy version of Arduino IDE v1.8 and the other is the new Arduino IDE v2.3.3. I have installed the SDuino using the IDE v1.8, and upon opening the IDE v2.3 the board seems to be reflected there too. 

Currently, I am using Windows 10, and I am using the new IDE to compile the code. Old IDE seems to be slow.

platima, 2025-05-16:
> https://github.com/akashkapashia/stm8_arduino/blob/master/README.md
> 
> use this meanwhile

Thank you!