tenbaht/sduino #159: Mere presence of (even unused) function with PinName argument makes DigitalWriteFast slower 10x

dvhx, 2024-04-11:
I'm using STM32F103C8T6 blue pill. I'm bitbangin DAC8830. Everything works fine, the output is sawtooth at around 4kHz. But if I add this function:

```
void asdf(PinName zz) {
  digitalWriteFast(zz, LOW);
}
```
Even though it's not even used, the speed drops to 500Hz which is almost 10x slowdown.

[fast.ino.txt](https://github.com/tenbaht/sduino/files/14948798/fast.ino.txt) - this is fast version
[slow.ino.txt](https://github.com/tenbaht/sduino/files/14948800/slow.ino.txt) - this is slow version
[image] - scope comparing fast and slow version

The only difference is really only the presence of one unused function. Of course in real code this function is used and does something but I was able to minimize the code to still cause the slowdown.

VimalO2, 2024-04-18:
can you provide the circuit connectiion

dvhx, 2024-04-18:
[image]