tenbaht/sduino #163: Installing issue in Arduino board manager!

bullygram, 2024-06-10:
Team I am unable to install the STMDuino, any help will be greatly appreciated!

`Tool sduino:STM8Tools@2019.02.05 already installed
Tool sduino:sdcc@build.11242 already installed
Downloading packages
sduino:stm8@0.5
Installing platform sduino:stm8@0.5
Failed to install platform: sduino:stm8.
Error: 13 INTERNAL: Cannot install platform: installing platform sduino:stm8@0.5: searching package root dir: no unique root dir in archive, found 'C:\Users\MySuerName\AppData\Local\Arduino15\tmp\package-1130421195\STM8S_StdPeriph_Driver' and 'C:\Users\MySuerName\AppData\Local\Arduino15\tmp\package-1130421195\cores'
`

bullygram, 2024-06-11:
Got it fixed, to install the board, use a portable version of Arduino IDE v1.8, and open the latest version of the IDE, the STM8 board is reflected in the board manager.

closed 2024-06-11 by bullygram

dmikushin, 2024-08-14:
I confirm that installing with Arduino IDE 1.8 works, and then usable in Arduino IDE 2.1.1