tenbaht/sduino #165: Could not upload the code

Foad5757, 2024-09-09:
Determine FLASH area

Due to its file extension (or lack thereof), ".pio\build\stm8sblue\firmware.hex" is considered as INTEL HEX format!
8104 bytes at 0x8000... Tries exceeded
*** [upload] Error 4294967295
While uploading get this error

Murugesh-Hobbyist, 2024-11-09:
Determine FLASH area
Due to its file extension (or lack thereof), "C:\Users\murug\AppData\Local\Temp\arduino_build_715628/Blink.ino.hex" is considered as INTEL HEX format!
1823 bytes at 0x8000... Tries exceeded
An error occurred while uploading the sketch

prosper00, 2024-11-09:
what board? what programmer? (you know the usb port on the 'blue' doesnt actually do anything other than power?)

Murugesh-Hobbyist, 2024-11-10:
Im using stm8s103f3p6 with ST-LINK V2 please have my settings.

[image]
[image]
[image]

prosper00, 2024-11-10:
hmm, looks right to me. Only thing I can think of is that the stlink may be a dud. I've had a few of those knockoff units fail to work with stm8. No idea why, but the quality and consistency is poor. Some have clone chips, some have stm32 variants that don't even (officially) even have USB (STM32F100, iirc).

Murugesh-Hobbyist, 2024-11-10:
I've encountered an issue while trying to program my STM32 board using an ST-LINK V2. Here’s what I observed:

Successful Upload but No Execution on STM32:
The program uploads successfully, and I receive a success message, as shown in the screenshot below. However, the STM32 board doesn’t appear to be running the program after upload (the LED does not blink as expected).

ST-LINK Utility Verification:
I tested the ST-LINK hardware using the ST-LINK Utility following guidance from tutorials. In this utility, the connection to the STM32 board works well, as shown here:

The ST-LINK Utility is able to erase the flash and verify the program. Additionally, I noticed that when the STM32 board is connected via the utility, uploading through the IDE is blocked, which I assume is expected behavior.

Multiple Devices Tested:
I have two ST-LINK V2 devices, an STM32, and two STM8S103F3P6 boards. None of the STM8 boards respond after upload, and the STM32 board doesn't run the program even though it’s uploaded.

Could you please help me troubleshoot why the STM32 isn’t running the uploaded program and offer any insights on what might be preventing it from executing?

prosper00, 2024-11-10:
Theres a lot of possibilities, from missing or bad decoupling or a missing voltage on one of the supply pins. maybe an issue with nRst, or the boot0/boot1 pins... or a firmware issue like a misconfigured clock tree or maybe problems with an interrupt vector, or any of a number of things. all i can really suggest is to look up an stm32cubeide tutorial, maybe a blinkie or something, and follow it closely.